SebKuzminsky/pycam #109: Engraving with Pocketing doesn't work (0.6.2)

jayMcBee, 2018-01-09:
Enabling Pocketing ('holes') in the Engraving strategy causes toolpath generation to fail with the error message on the Console that openvoronoi library failed to load.

sumpfralle, 2018-01-11:
Indeed this is just an information (not an error). I clarified the wording now. Openvoronoi was not yet integrated, thus this warning does not really indicate a degraded behaviour.

What exactly is not working for you (besides the message)?

jayMcBee, 2018-01-11:
Well - no toolpath is generated..

sumpfralle, 2018-02-15:
Yes: sadly the pocketing algorithm of pycam is not stable - thus it fails for many (seemingly trivial) 2D shapes.

Maybe someone wants to work on the integration of an external pocketing library?

Possible candidates:
* opencamlib (#25)
* clipper (#26)

sumpfralle, 2018-02-15:
See #114.

closed 2018-02-15 by sumpfralle