SebKuzminsky/pycam #111: Failed to load openvoronoi library

rogeliomartin, 2018-02-06:
ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Plugins/Tasks.py", line 316, in generate_toolpath toolpath = task_resolver(task, callback=draw_callback) File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Plugins/TaskTypes.py", line 71, in run_task maxz=box.upper.z, draw_callback=callback)   File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/PathGenerators/EngraveCutter.py", line 39, in GenerateToolPath motion_grid = list(motion_grid)   File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Toolpath/MotionGrid.py", line 469, in get_lines_grid     callback=callback)   File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Toolpath/MotionGrid.py", line 527, in get_pocketing_polygons  poly = get_pocketing_polygons_simple(polygons, offset, pocketing_type, callback) File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Toolpath/MotionGrid.py", line 575, in get_pocketing_polygons_simple result = poly.get_offset_polygons(offset) File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Geometry/Polygon.py", line 959, in get_offset_polygons group.append(line)   File "/usr/local/lib/python2.7/dist-packages/pycam-0.6.2-py2.7.egg/pycam/Geometry/Polygon.py", line 234, in append  raise ValueError("This line does not fit to the polygon") ValueError: This line does not fit to the polygon

unrealsolver, 2018-02-08:
@rogeliomartin , how it's related to openvoronoi? Could you please provide steps to reproduce the bug and also the file (if possible). This will be really helpful.

rogeliomartin, 2018-02-10:
hi,

the error just pop out

1.- load model

2.- calculate the 2D projection

3.- togle directions end revise directions

4.-set operation strategy engraving make the selection trace model the 2D model projected milling style climb / down , pocketing (Material) , checked radius compensation

5.-setup bounds as loaded

6.- task (select task#1)
7.-check colision model (model 1)
8.- click on generate toolpath and then the error pops out

sumpfralle, 2018-02-15:
@rogeliomartin: I am sorry, but this looks like another instance of the fragile polygon offsetting code in pycam failing.

See #114.

closed 2018-02-15 by sumpfralle