SebKuzminsky/pycam #113: External polygon offsetting library: svgpathtools

sumpfralle, 2018-02-15:
Maybe svgpathtools could be suitable external library for calculating polygon offsets ([proposed by SebKuzminsky](https://github.com/SebKuzminsky/pycam/issues/79#issuecomment-343781228)):

> https://github.com/mathandy/svgpathtools
>
> I don't yet know if svgpathtools is the external libaray we should use. If it is, i think we have two options:
>
>    Package svgpathtools as a debian package and have pycam Depend on it, or
>    Include svgpathtools as a submodule in the pycam git repo and include it in the pycam package.

Is it suitable for polygon offsetting? Should we use it?

SebKuzminsky, 2018-02-16:
svgpathtools does not currently include polygon offsetting, though it does include primitives from which polygon offsetting can be implemented.  I've sketched out a polygon offsetting function based on svgpathtools, but bugs in svgpathtools prevent it from working robustly.  Details in #114.

ebo, 2018-02-16:
On Feb 16 2018 10:05 AM, Sebastian Kuzminsky wrote:
> svgpathtools does not currently include polygon offsetting, though it
> does include primitives from which polygon offsetting can be
> implemented.  I've sketched out a polygon offsetting function based 
> on
> svgpathtools, but bugs in svgpathtools prevent it from working
> robustly.  Details in #114.

by polygon offsetting I assume you mean the 2D equivalent of offset 
surfaces or parallel curve 
<https://en.wikipedia.org/wiki/Parallel_curve>.  Am I correct?

As a note, I worked with the Spatial Ontology Community of Practice 
(SOCoP) and GeoVoCamp a couple of years ago working on formal 
definitions of map contours.  We had talked about extending this to 
surfaces, and abstract surfaces over time, but I cannot remember where 
that got to.  If interested I will look for the paper that Torsten wrote 
from that ontological formalism.  The proofs will provide you with some 
fodder for unit tests and regression tests.

SebKuzminsky, 2018-02-16:
@ebo Yes exactly, the offset polygon consists of the parallel curve (with a specific distance) of the original polygon.  It's used for tool radius compensation, and it can also be used to compute a pocketing toolpath.

ebo, 2018-02-16:
@Sebastian, yep.  I have used it before.  Let me know off list and I 
will see what I can get for you for theoretical references that might be 
very helpful (like Ferin's work on curves and surfaces)

On Feb 16 2018 12:07 PM, Sebastian Kuzminsky wrote:
> @ebo Yes exactly, the offset polygon consists of the parallel curve
> (with a specific distance) of the original polygon.  It's used for
> tool radius compensation, and it can also be used to compute a
> pocketing toolpath.

SebKuzminsky, 2018-02-16:
One thing I like about svgpathtools is that it supports all the path segment types of SVG: lines, elliptic arcs, and quadratic and cubic bezier splines.  Arcs especially are common in part designs, and supporting them without using linear-spline approximations is a good thing.

ebo, 2018-02-16:
Nice!  What are the bugs that are keeping it from being used?  From a 
comment before I thought that svgpathtools was sufficiently buggy that 
you could not use it.

On Feb 16 2018 12:51 PM, Sebastian Kuzminsky wrote:
> One thing I like about svgpathtools is that it supports all the path
> segment types of SVG: lines, elliptic arcs, and quadratic and cubic
> bezier splines.  Arcs especially are common in part designs, and
> supporting them without using linear-spline approximations is a good
> thing.

SebKuzminsky, 2018-02-16:
The bugs i've found in svgpathtools (so far...) are:
* mathandy/svgpathtools#35
* mathandy/svgpathtools#37
* mathandy/svgpathtools#41
* mathandy/svgpathtools#43

I fixed some of those issues and offered PRs, but they have not been accepted yet:
* mathandy/svgpathtools#40
* mathandy/svgpathtools#42

The last bug, mathandy/svgpathtools#43, is the only thing I know of preventing the svgpathtools-based polygon offsetting code I wrote from working reliably.

ebo, 2018-02-16:
I'm swamped enough I will unfortunately not be able to look into this.

On Feb 16 2018 1:09 PM, Sebastian Kuzminsky wrote:
> The bugs i've found in svgpathtools (so far...) are:
> * mathandy/svgpathtools#35
> * mathandy/svgpathtools#37
> * mathandy/svgpathtools#41
> * mathandy/svgpathtools#43
>
> I fixed some of those issues and [...]

sumpfralle, 2018-08-20:
Just updating the status of the of the open issues in svgpathtools (based on the list by @SebKuzminsky):
* mathandy/svgpathtools#35
* mathandy/svgpathtools#37
* mathandy/svgpathtools#43