SebKuzminsky/pycam #114: Polygon offsetting / pocketing fails

sumpfralle, 2018-02-15:
The self-made polygon offsetting algorithm in pycam is fragile und fails to work even with seemingly trivial polygons.

Some problems of the algorithm are discussed in #79.

The offsetting algorithm fails with the following symptoms:
* error message `ValueError: This line does not fit to the polygon` - #111
* empty result (without error messages) - #109
* corners are mirrored inside out - #79
* ...

The polygon offsetting algorithm in pycam is broken beyond repair. We need to replace it with something stable.

The following external libraries could provide an alternative:
* opencamlib - #25 
* clipper - #26
* svgpathtools - #113

I propose to use this issue / thread in the following way:
* Feel free to share (briefly) more symptoms or error messages related to a failed polygon offsetting operation. This useful for search engines and may lead users facing this problem to this ticket. Please do not expect that someone replies or fixes your issue (see above for the reasoning).
* You are invited to refer to other possible external libraries for polygon offsetting.
* If you really feel that you need to discuss improvements of the current offsetting code in pycam, then please open a separate ticket and share your progress while working on this topic there. Currently no one volunteered to work on this - probably because it feels like a waste of effort.

SebKuzminsky, 2018-02-16:
I've spent the past several weeks playing with https://github.com/mathandy/svgpathtools.

It's a pure python module that reads SVG files and parses the paths into an internal representation.  The path itself and the path elements (lines, elliptic arcs, and quadratic & cubic splines) are all represented by python classes with many geometric primitives defined on them.

There is currently no path offsetting function, but the required primitives are there, and some sample code is provided that implements the beginning of a path offsetting function.  I extended that example code into a rough implementation of Chen & McCains algorithm "[Polygon Offsetting by Computing Winding Numbers](http://www.me.berkeley.edu/~mcmains/pubs/DAC05OffsetPolygon.pdf)".

This exposed several bugs in svgpathtools (mathandy/svgpathtools#35, mathandy/svgpathtools#37, mathandy/svgpathtools#41, and mathandy/svgpathtools#43).  I fixed some but not all of the bugs (mathandy/svgpathtools#40, mathandy/svgpathtools#42), but the fixes have not been accepted (yet?).

The result so far is a robust path offsetting function, and a simple pocketing function based on it.  The only remaining flaw I know of is a bug in Arc/Arc intersections (mathandy/svgpathtools#43) that often crashes the offsetting code.

The offsetting algorithm isn't tied to svgpathtools, and we may be better off using a different geometry library, hopefully one that's more actively maintained.

[image]
[image]
[image]
[image]
[image]

jf---, 2018-02-28:
duly impressive work, what a cool reference to `Chen & McCains algorithm`

SebKuzminsky, 2018-02-28:
The pocketing is implemented in a program called `svg2gcode`, in this repo:  https://github.com/SebKuzminsky/svg2gcode

It's super quick and hacky, and not really intended for anything other than me playing with svgpathtools  and testing the offsetting algorithm.  You've been warned!