SebKuzminsky/pycam #115: UnboundLocalError

valeriob01, 2018-02-16:
Enabled 4 parallel local processes
Storing a state revision (1/20)

(run_gui.py:31926): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:31926): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed
Font directory: /home/sel/Downloads/newpycam/pycam/share/fonts
Loading workspace from file: /home/sel/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
Imported STL model: 12 triangles
Imported 1 items into 'export_settings'
Storing a state revision (2/20)

(run_gui.py:31926): Gtk-WARNING **: Allocating size to GtkBox 0x564c14dd22a0 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?
Storing a state revision (3/20)

(run_gui.py:31926): Gtk-WARNING **: Allocating size to GtkBox 0x564c14c97f50 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?

(run_gui.py:31926): Gtk-WARNING **: Allocating size to GtkNotebook 0x564c14332dd0 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?
Traceback (most recent call last):
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/Clipboard.py", line 86, in _update_clipboard_widget
    models = self._get_exportable_models()
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/Clipboard.py", line 78, in _get_exportable_models
    models = self.core.get("models").get_selected()
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/__init__.py", line 427, in get_selected
    return self._get_gtk_selected(**kwargs)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/__init__.py", line 458, in _get_gtk_selected
    if (selection_mode == self._gtk.SelectionMode.MULTIPLE) or force_list:
UnboundLocalError: local variable 'selection_mode' referenced before assignment

sumpfralle, 2018-02-17:
wow - I am impressed, that you were able to trigger this one so quickly!
I thought, it would be a rare race condition during the shutdown of the program.

closed 2018-02-17 by sumpfralle

valeriob01, 2018-02-17:
It just popped out alone after an update.

I have a script that updates and launches the program, I wrote it to be
more efficient in testing

commit 46375d56 (2018-02-17): fix UnboundLocalError during shutdown

Closes: #115