SebKuzminsky/pycam #116: dxf polyline error

stevanov, 2018-03-18:
Just tried Pycam for the first time, imported one of my (dxf) drawings and got the following error. It said to pass it on to you guys. I'm using pycam installed trough yaourt, 0.6.2
If any more info is needed i'm happy to supply it

```
16:53 INFO Enabled 4 parallel local processes
16:53 INFO Font directory: /usr/share/pycam/share/fonts
16:53 INFO Imported STL model: 0 vertices, 0 edges, 1444 triangles
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/courier.cxf': 122 letters
16:54 INFO *** skipped 1 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/cyrillic_ii.cxf'. Reason: Failed to parse unicode character at line 539.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/gothgbt.cxf': 93 letters
16:54 INFO *** skipped 14 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/normallatin2.cxf'. Reason: Failed to parse unicode character at line 553.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/romanc.cxf': 189 letters
16:54 INFO *** skipped 1 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/romand.cxf'. Reason: Failed to parse unicode character at line 2421.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/romanp.cxf': 189 letters
16:54 INFO *** skipped 5 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/standard.cxf'. Reason: Failed to parse unicode character at line 638.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/symbol.cxf': 103 letters
16:54 INFO *** skipped 4 similar message(s) ***
16:54 ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/usr/share/pycam/pycam/Gui/Project.py", line 386, in gui_activity_guard_wrapper     result = func(self, *args, **kwargs)   File "/usr/share/pycam/pycam/Gui/Project.py",16:53 INFO Enabled 4 parallel local processes
16:53 INFO Font directory: /usr/share/pycam/share/fonts
16:53 INFO Imported STL model: 0 vertices, 0 edges, 1444 triangles
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/courier.cxf': 122 letters
16:54 INFO *** skipped 1 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/cyrillic_ii.cxf'. Reason: Failed to parse unicode character at line 539.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/gothgbt.cxf': 93 letters
16:54 INFO *** skipped 14 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/normallatin2.cxf'. Reason: Failed to parse unicode character at line 553.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/romanc.cxf': 189 letters
16:54 INFO *** skipped 1 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/romand.cxf'. Reason: Failed to parse unicode character at line 2421.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/romanp.cxf': 189 letters
16:54 INFO *** skipped 5 similar message(s) ***
16:54 WARNING CFXImporter: Skipped font defintion file '/usr/share/pycam/share/fonts/standard.cxf'. Reason: Failed to parse unicode character at line 638.
16:54 INFO CXFImporter: Imported CXF font from '/usr/share/pycam/share/fonts/symbol.cxf': 103 letters
16:54 INFO *** skipped 4 similar message(s) ***
16:54 ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/usr/share/pycam/pycam/Gui/Project.py", line 386, in gui_activity_guard_wrapper     result = func(self, *args, **kwargs)   File "/usr/share/pycam/pycam/Gui/Project.py", line 534, in load_model_file     callback=progress.update)   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 881, in import_model     callback=callback)   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 118, in __init__     self.parse_content()   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 241, in parse_content     self.parse_lwpolyline()   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 410, in parse_lwpolyline     self.lines.extend(pycam.Geometry.get_bezier_lines( AttributeError: 'module' object has no attribute 'get_bezier_lines' line 534, in load_model_file     callback=progress.update)   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 881, in import_model     callback=callback)   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 118, in __init__     self.parse_content()   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 241, in parse_content     self.parse_lwpolyline()   File "/usr/share/pycam/pycam/Importers/DXFImporter.py", line 410, in parse_lwpolyline     self.lines.extend(pycam.Geometry.get_bezier_lines( AttributeError: 'module' object has no attribute 'get_bezier_lines'
```

closed 2018-03-18 by sumpfralle

stevanov, 2018-03-18:
awesome thanks! @sumpfralle

sumpfralle, 2018-03-18:
Thank you for your report!

Would it be easy for you to share your DXF file (maybe minimize it for the bezier lines)?
I would like to add it to the repository for a test.

stevanov, 2018-03-18:
Yeah sure, here you go!
[bezier-errorfile.zip](https://github.com/SebKuzminsky/pycam/files/1823087/bezier-errorfile.zip)
I stripped it of most details but this is the part of the drawing pycam trips over. Hope this helps

sumpfralle, 2018-03-19:
Thank you for the file!
I added it to the tests and also fixed a bit more issues with regards to bezier lines.

stevanov, 2018-03-21:
great, happy to help, i'll be sure to report if i run into anything else

commit 41544754 (2018-03-18): DXFImporter: fix import of "get_bezier_lines"

Closes: #116
commit 12b97187 (2018-03-18): DXFImporter: fix import of "get_bezier_lines"

Closes: #116
(cherry picked from commit 415447545c63e338d01b2a3e305a94f73ca5f5a4)