SebKuzminsky/pycam #118: Import from SVG fails: undefined in DELAYBIND

leonerd, 2018-04-17:
Attempting to import an SVG file, I get the following error:

```
Successfully converted SVG file to EPS file
SVGImporter: failed to convert EPS file (/tmp/tmpVb3Mw7.eps) to DXF file (/tmp/tmpJqJDE0.dxf): pstoedit: version 3.70 / DLL interface 108 (built: Feb 19 2018 - release build - g++ 7.3.0 - 64-bit) : Copyright (C) 1993 - 2014 Wolfgang Glunz

 *** WARNING - the DELAYBIND command has been deprecated, and will be
               removed in the next version. For now you can restore
               the behaviour by using -dREALLYDEALYBIND but if you
               require continued use of this command you should contact
               the Ghostscript developers. Commercial customers of
               Artifex should email their support contact, free users
               are encouraged to talk to us on the #ghostscript IRC
               channel on irc.freenode.net.

Unrecoverable error: undefined in DELAYBIND
PostScript/PDF Interpreter finished. Return status 65280 executed command : /usr/bin/gs -q -dDELAYBIND -dWRITESYSTEMDICT -dNODISPLAY -dNOEPS "/tmp/psinB0W3vm"
The interpreter seems to have failed, cannot proceed !
```

leonerd, 2018-04-17:
In case it's useful, the `.svg` file is attached here (named `.svg.txt` so github is happy with it)
[front.svg.txt](https://github.com/SebKuzminsky/pycam/files/1917801/front.svg.txt)

SebKuzminsky, 2018-04-26:
I agree this looks like a bug in ghostscript or pstoedit, not in PyCAM (though obviously pycam is affected by it, since it uses those programs).

In the long run we should teach PyCAM to read SVGs directly (#119).

commit f29c90a5 (2018-08-20): Implement direct SVG import without external programs

Previously inkscape and pstoedit were chained in order to convert SVG
files to DXF and import the result. This process turned out to be error
prone (besides requiring additional dependencies):
* https://github.com/SebKuzminsky/pycam/issues/118

The new implementation uses svg.path for interpreting basic SVG paths
directly in Python without calling external commands.
For now only a few SVG entities are supported (paths and rectangles).
More entities will probably follow (on demand), like circles and
simple transformations. Full SVG support  [... cut]
commit 215e948a (2019-07-28): Implement direct SVG import without external programs

Previously inkscape and pstoedit were chained in order to convert SVG
files to DXF and import the result. This process turned out to be error
prone (besides requiring additional dependencies):
* https://github.com/SebKuzminsky/pycam/issues/118

The new implementation uses svg.path for interpreting basic SVG paths
directly in Python without calling external commands.
For now only a few SVG entities are supported (paths and rectangles).
More entities will probably follow (on demand), like circles and
simple transformations. Full SVG support  [... cut]
closed 2018-08-20 by sumpfralle