SebKuzminsky/pycam #120: Operate only on specific selected features of a model

SebKuzminsky, 2018-04-26:
It is a common CAM use-case to import a complete drawing of a part and do different operations on different parts of the model.  I think there's currently no convenient way to do that in PyCAM.

For example:
1. Helix-mill through-holes in the six small circles (bolt holes).
2. Machine partial-depth pockets in the four large circles (counter-bores for bolt heads).
3. Full depth cut around the outline (with holding tabs!) to free the part from the raw material.  
[image]

DXF file renamed to .txt to placate github: [geiger_top.dxf.txt](https://github.com/SebKuzminsky/pycam/files/1952134/geiger_top.dxf.txt)

sumpfralle, 2018-04-26:
I could imagine, that we can split 2D models into polygons (and allow to join some of them again into a single model for convenience).

This could get a bit complicated for big models (with many polygons). Right now, I have no specific idea, how this could be solved. Maybe grouping by size, position or shape?

Would this be a step forward in the direction, that you are imagining?

SebKuzminsky, 2018-04-26:
Each of the 11 closed paths in the picture above is a separate entity in the DXF file, and SVG files similarly encode separate paths as separate objects.
I'm imagining something like the GUI in [CamBam](http://www.cambam.info/) (which is closed source, boo hoo) and [dxf2gcode](https://sourceforge.net/projects/dxf2gcode/) (which is open source, yay).  In both of those you can click on a paths to select them, and generate machining operations that only apply to the selected paths.
I don't know how this would for for a tesselated solid, since they have no notion of selectable features...

SebKuzminsky, 2018-04-26:
Here are some videos showing off those GUIs.

dxf2gcode:
* https://www.youtube.com/watch?v=RR1ujV8BoYM&feature=youtu.be&t=375 (LOUD!)
* https://www.youtube.com/watch?v=mnPhRSFKPJE&t=330

CamBam:
* https://www.youtube.com/watch?v=RFBE2RuNCD8

openbuilds-engineer, 2018-04-27:
(; also checkout cam.openbuilds.com