SebKuzminsky/pycam #121: The main GUI window locks up while generating a toolpath

lgeek, 2018-07-21:
The main GUI window locks up while generating a toolpath and no content is rendered in that window. I'm running the master branch (commit d0d0091014f518e6). I've previously used PyCAM 0.6.2 from the Arch Linux package, which was periodically updating that window with the toolpath generation status.

sumpfralle, 2018-07-21:
Could you please save your workspace settings and share them with us? (including models, if necessary)
Then I could take a look how where it hangs.

sumpfralle, 2018-08-18:
Ping?

lgeek, 2018-08-18:
Sorry, I can't share these particular models. But the GUI thread always seems to be blocked for a large part of the time it takes to generate the toolpath. I'll prepare a different example if it's not clear what I mean.

sumpfralle, 2018-08-18:
Maybe just tell us, what is written in the progress bar, while the GUI is unresponsive. Thank you!

ByteHamster, 2021-01-10:
For me, the GUI becomes unresponsive on master 4687f2f62bd75078a7b1b70339e9a5df75b64c63 when using the "surface" operation. It looks like DropCutter executes the process line-by-line. Because surfacing seems to be just one single line, the process hangs until the operation is finished. On my device, this can take around 20 minutes of the application being completely unresponsive (black window). In a task manager, I can see that pycam starts 8 threads but only one of them actually does something.

I think the problem is that `get_toolpath` is called on the main rendering thread for both OpenGL window and main window. That method blocks until the toolpath is generated. After the initial generation, it is cached and returned much faster.

My idea for a fix would be to add a method that can check if the toolpath is generated or not. If it isn't, skip the `get_toolpath` method on the main thread. The "Generate toolpath" button still executes the generation on the UI thread, though. I tried starting a thread for that but that resulted in segfaults that I did not understand. Unfortunately, I am not experienced enough in Python. @sumpfralle If I was to create a PR, could you please have a look and see if you have an idea? Looks like the development is not too active recently, so before preparing a PR, it would be good to know if it will be reviewed :)

selroc, 2021-01-10:
I have the feeling that we have to fix things ourselves, in the last two years @sumpfralle has been unresponsive himself and for the most part, asking for contributed fixes instead of fixing things himself.

ByteHamster, 2021-02-03:
Sorry, I switched to BlenderCam, so I will probably not attempt to fix it anytime soon. The thing I like about BlenderCam is that you can modify the path visually if you need to. It's harder to use than pycam but it scratches my itch for now.