SebKuzminsky/pycam #122: Slice removal doesnt work

mirecta, 2018-07-26:
i run fresh checkout with default object (cube) defauld bounds , and generate toolpath quick removal and path remove everything (cube too :D) if i choose slice removal type spiral , it is still grid

sumpfralle, 2018-07-26:
Thank you for your report!

Are these two distinct issues, you are referring to? In this case: please open another issue.

In order to reproduce your issue, you need to share your model and your workspace settings file with us. Additionally you need to tell us the version of pycam you are using. Thanks!

mirecta, 2018-07-27:
i dont touch anything i run program actual HEAD commit d0d0091014f518e6d193ac153028967a786fd3f9
and then i click Genetate toolpath
[image]
[image]
[image]
[image]
[image]
[image]

mirecta, 2018-07-27:
as u can see on last image tool 6mm dia with this path remove everything

sumpfralle, 2018-07-28:
Indeed the visualization was not correct - e.g. it lacked the tool path moves related to the safety height. Thus the visualization was misleading.
The export to a file was nevertheless correct.

Thank you for your report!

commit 96b39375 (2018-07-28): Toolpath view: use export filters (e.g. safety height)

Previously only tool-based toolpath settings were applied. Thus import
settings like "safety height" were not applied in the visualization (but
in the file export).

Closes: #122
closed 2018-07-28 by sumpfralle