SebKuzminsky/pycam #124: python-rsvg dependency problem on ubuntu 18.04

pedrosacramento, 2018-08-07:
The dependency package python-rsvg is not available on ubuntu 18.04. It was removed as stated here (if I understood well): https://bugs.launchpad.net/ubuntu/+source/gnome-python-desktop/+bug/1739797

sumpfralle, 2018-08-07:
I think, we can silently drop this dependency. It is only used in the v0.6 release branch and its only purpose is the visualization of some icons.

@SebKuzminsky: what do you think?

craig9, 2019-03-08:
I've just ran into this issue too, the requirements page for pycam still asks for python-rsvg. Could it perhaps be removed?

Also python-guppy on 18.10 seems to be available via pip, so the instructions there may need tweaking too.

sumpfralle, 2019-03-09:
> I've just ran into this issue too, the requirements page for pycam still asks for python-rsvg. Could it perhaps be removed?

I did this now in 146c7389127eb394345af1227251e04e1fc7874f.

> Also python-guppy on 18.10 seems to be available via pip, so the instructions there may need tweaking too.

I also removed this dependency (94af38cb8c32e6c872e7c9161a31d6ef0e8d9b4e), since it is just an optional debugging helper.

> Finally, python-opengl, python-gtk2 and python-gtkglext1 seem to be requirements too.

Out of these only "python-opengl" can be installed via pypi (it is listed there as `OpenGL`). The other packages seem to be only available via the system's package managers. Thus they cannot be mentioned in `requirements.txt`. Or did you want to refer to something else?

craig9, 2019-03-09:
Thanks for the changes Lars. I can now see that those three packages are in fact mentioned on this web page: http://pycam.sourceforge.net/requirements/ - which is the requirements page I was referring to initially. I must have missed those entries.

As an aside, is the sourceforge website the main website? Or should it point to the github page now?

The git hub requirements page has this install command for debian/ubuntu requirements:

$ apt-get install python3-gi python3-opengl gir1.2-gtk-3.0

And the sourceforge requirements page has this install command for ubuntu:

$ sudo apt-get install python-gtk2 python-opengl python-gtkglext1 python-rsvg python-guppy

sumpfralle, 2019-03-09:
> As an aside, is the sourceforge website the main website?

Yes, indeed. I just forgot to upload the updated documentation recently. Now it is up to date again. Thank you for the reminder.

closed 2019-03-09 by sumpfralle