SebKuzminsky/pycam #127: Please report this to developers ....

masterleo, 2018-09-09:
after importing a light file, and preparing it, I got this error. 
Imported STL model: 152 vertices, 0 edges, 283 triangles
An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot!```

Traceback (most recent call last):
  File "/home/AvilabPrinter/Apps/pycam-0.5.1/src/pycam/Gui/Project.py", line 4041, in generate_toolpath
    self.model, toolpath_settings, callback=draw_callback)
  File "/home/AvilabPrinter/Apps/pycam-0.5.1/src/pycam/Toolpath/Generator.py", line 62, in generate_toolpath_from_settings
    backend, callback)
  File "/home/AvilabPrinter/Apps/pycam-0.5.1/src/pycam/Toolpath/Generator.py", line 354, in generate_toolpath
    callback)
  File "/home/AvilabPrinter/Apps/pycam-0.5.1/src/pycam/PathGenerators/DropCutter.py", line 109, in GenerateToolPath
    callback=progress_counter.update):
  File "/home/AvilabPrinter/Apps/pycam-0.5.1/src/pycam/Utils/threading.py", line 649, in run_in_parallel_local
    for result in imap_func(func, args):
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 668, in next
    raise value
AttributeError: 'NoneType' object has no attribute 'add'

```

My client is vaiting, so please make pycam work ! Also with large STLs ;)

masterleo, 2018-09-09:
My environment  is like this : 
[image]

sumpfralle, 2018-09-09:
This seems to be a duplicate of #57.

Maybe disabling *parallel processing* (`Settings -> Parallel Processing`) helps?

closed 2018-09-09 by sumpfralle