SebKuzminsky/pycam #129: Error while importing SVG.

hugovangalen, 2018-10-15:
As the error message requested the traceback to be sent to the developers, here you go. :)

```
Traceback (most recent call last):
  File "/data/apps/pycam/pycam/Gui/Project.py", line 84, in gui_activity_guard_wrapper
    result = func(self, *args, **kwargs)
  File "/data/apps/pycam/pycam/Gui/Project.py", line 545, in load_model_file
    self.settings.get("models").add_model(model_params, name=name_suggestion)
  File "/data/apps/pycam/pycam/Plugins/Models.py", line 167, in add_model
    new_model.set_application_value("visible", True)
  File "/usr/lib/python3.5/contextlib.py", line 66, in __exit__
    next(self.gen)
  File "/data/apps/pycam/pycam/Flow/history.py", line 118, in merge_history_and_block_events
    yield
  File "/usr/lib/python3.5/contextlib.py", line 66, in __exit__
    next(self.gen)
  File "/data/apps/pycam/pycam/Utils/events.py", line 141, in blocked_events
    self.emit_event(event)
  File "/data/apps/pycam/pycam/Utils/events.py", line 101, in emit_event
    handler.func(*(handler.args + args), **kwargs)
  File "/data/apps/pycam/pycam/Plugins/__init__.py", line 171, in <lambda>
    result = lambda *args: self.core.emit_event(func, *params)
  File "/data/apps/pycam/pycam/Utils/events.py", line 101, in emit_event
    handler.func(*(handler.args + args), **kwargs)
  File "/data/apps/pycam/pycam/Plugins/OpenGLViewDimension.py", line 51, in update_model_dimensions
    models = [m.get_model() for m in self.core.get("models").get_visible()]
  File "/data/apps/pycam/pycam/Plugins/OpenGLViewDimension.py", line 51, in <listcomp>
    models = [m.get_model() for m in self.core.get("models").get_visible()]
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 292, in wrapped
    return self.get_cached(inst, args, kwargs, calc_function)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 340, in get_cached
    cache_item = CacheItem(time.time(), calc_function(inst, *args, **kwargs))
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 1004, in get_model
    model = self.get_value("source").get(CollectionName.MODELS)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 292, in wrapped
    return self.get_cached(inst, args, kwargs, calc_function)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 340, in get_cached
    cache_item = CacheItem(time.time(), calc_function(inst, *args, **kwargs))
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 719, in get
    return self._get_source_location(source_type)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 256, in inner_function
    return func(self, *args, **kwargs)
  File "/data/apps/pycam/pycam/workspace/data_models.py", line 760, in _get_source_location
    return detected_filetype.importer(detected_filetype.uri)
  File "/data/apps/pycam/pycam/Importers/SVGDirectImporter.py", line 249, in import_model
    model.append(polygon)
  File "/data/apps/pycam/pycam/Geometry/Model.py", line 407, in append
    super().append(item)
  File "/data/apps/pycam/pycam/Geometry/Model.py", line 151, in append
    self._update_limits(item)
  File "/data/apps/pycam/pycam/Geometry/Model.py", line 143, in _update_limits
    self.minx = min(self.minx, item.minx)
TypeError: unorderable types: NoneType() < float()
```

sumpfralle, 2018-10-16:
Thank you for your report!

I think, I fixed the issue with e935650bc370a685b05024bf63ba83249f27bb38.
Please report back, if the import works for you now.

hugovangalen, 2018-10-19:
Thanks, that seems to fix it! 

I cannot confirm it did it correctly though, because I am hit by issue #97 :(

sumpfralle, 2018-10-20:
What a pity :(
But at least the importer is better now ...

closed 2018-10-20 by sumpfralle