SebKuzminsky/pycam #131: Waterline problem

burbilog, 2018-10-21:
If I create a process with waterline strategy and generate toolpath then it generates moves through everything, like there is no model at all:

[image]

If I switch strategy to surfacing, change milling style to "conventional/up" and then switch strategy back to waterline (milling style is NOT available in waterline strategy) then waterline toolpath is going to be generated correctly:

[image]

I found hint to change milling style here: https://sourceforge.net/p/pycam/bugs/76/ so this bug is VERY old.

The trouble is, waterline dialog has no miling style control and no sane default. Please, make that milling style control available in waterline strategy and make conventional/up default. It took a lot of time to find out what's wrong with waterline...

sumpfralle, 2018-10-21:
I would like to reproduce it. Which version do you use?

burbilog, 2018-10-22:
%git log|head                                                                                                                                             
    commit e935650bc370a685b05024bf63ba83249f27bb38   
    Author: Lars Kruse <devel@sumpfralle.de>
    Date:   Tue Oct 16 05:14:52 2018 +0200

        SVGDirectImporter: skip empty polygons

I just reproduced it on another computer with the same OS Kubuntu 18.04.1 LTS. How to reproduce:
1. git clone https://github.com/SebKuzminsky/pycam.git
2. run pycam (I have to run it as MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5" python3 pycam/run_gui.py otherwise it does not work)
3. go to Process tab
4. select Surface (fine)
5. change strategy to waterline (don't do anything else there)
6. go to Tasks tab
7. click Finishing
8. click Generate Toolpath 

and that's it: https://imgur.com/a/zRGFoSy (sorry for external image hosting, can't upload picture to github, it says something went wrong).

sumpfralle, 2018-10-25:
Thank you for this detailed bug report for such an interesting behaviour!
It should be fixed now - even though there is currently no way to specify the milling style (which would require a good amount of changes in order to allow only the allowed subset of usable milling styles).

commit befb414b (2018-10-25): waterline: enforce a usable milling style for calculating the motion grid

The "IGNORE" milling style breaks the calculation of the (hacky)
waterline PathGenerator, since this generator relies on all path
segments running along the same direction (not back and forth).

Closes: #131
closed 2018-10-25 by sumpfralle