SebKuzminsky/pycam #133: Sourceforge standalone windows executable downloads link broken

ftc, 2018-12-16:
On the installation page, the link to the windows standalone executable is broken. As well as all links that point to "http://pycam.sourceforge.net/download.html".

Not a blocker as I can install the development version but would be a nice thing to have working.

Overdrivr, 2019-06-28:
Noticed this too, where are the instructions for building from source on Windows ?

sumpfralle, 2019-06-28:
Where did you find a link pointing at http://pycam.sourceforge.net/download.html?

I am only aware of the download link in http://pycam.sourceforge.net/ pointing at https://sourceforge.net/projects/pycam/files/pycam/. There you can find the old windows standalone executable.

Overdrivr, 2019-06-29:
The links are at this page : http://pycam.sourceforge.net/installation/