SebKuzminsky/pycam #134: OpenGL visualization requires legacy features (before 3.0)

sumpfralle, 2018-12-16:
Sadly pycam currently uses many [OpenGL legacy](https://www.khronos.org/opengl/wiki/Legacy_OpenGL) functions (removed from the 3.0 specification).

Thus modern graphic cards fail to work with with `invalid operation` errors at various places in the code (usually at least for `glMatrixMode`).

This breaks the 3D visualization for many (and soon: most) systems.

sumpfralle, 2018-12-16:
The migration of the legacy functions to the new API seems to be non-trivial.

Personally I have no interest in gaining OpenGL-related knowledge. Thus the OpenGL visualization will need to be removed, unless someone contributes this migration.

I researched possible alternatives (high-level 3D visualization toolkits or 3D exchange formats). For now I am favoring an export to X3D (see #135).

lava, 2019-03-31:
Note that *most* modern graphics cards (in particular all Nvidia and AMD cards) still support legacy OpenGL, e.g. the famous `glxgears` test program is still written using the legacy API and works basically everywhere I tried.

The reason why it doesn't work for PyCAM in particular is that Gdk is by default requesting the legacy features to be turned off for the OpenGL contexts it creates. After applying #140, I'm able to run the OpenGL code unchanged without any errors.

sumpfralle, 2019-03-31:
Thank you for the clarification!