SebKuzminsky/pycam #135: Visualization via X3D

sumpfralle, 2018-12-16:
Due the broken state of the current OpenGL visualization (see #134), it would be good to find another method for visualizing the current model and toolpaths.

Maybe [X3D](https://en.wikipedia.org/wiki/X3D) would be a good candidate.

It could work like this:
* pycam creates an X3D export whenever models or toolpaths change
* the X3D export is rendered in a [web browser widget](https://packages.debian.org/libwebkit2gtk)
    * emergency fallback: open in a separate browser

This will also be useful in case anyone wants to implement a web-based application around pycam's scripting feature.

valeriob01, 2018-12-17:
I'm sorry to hear you want to use a web browser widget or a separate browser. I'm sure there can be a better solution.

skrewz, 2018-12-18:
I wouldn't mind a web browser (widget) based approach, FWIW. But isn't the point also that once expressed in X3D, the choices for a rendering platform are open?

valeriob01, 2018-12-18:
To me the web-widget is an ugly approach and a crude hack.

sumpfralle, 2018-12-19:
> the choices for a rendering platform are open?

Yes. At least in Debian I see [view3dscene](https://packages.debian.org/view3dscene), but it is probably not possible to control this rendering externally. Thus the javascript-based rendering via [x3dom](http://x3dom.org/) looks like my favorite option for now.

> To me the web-widget is an ugly approach and a crude hack.

I would be happy, if someone would take over the maintenance of the current OpenGL legacy code in pycam and port it to the current OpenGL standards: 3.x (since 2008) or maybe even 4.x (2010). I am quite sure, that no new graphics hardware sold today is able to render the OpenGL commands in pycam (targeting 2.x, 2004) anymore. Thus a change is inevitable. Just the direction of progress is open for discussion. I welcome any input.

sumpfralle, 2018-12-20:
Just for the curious:

[image]

Up to now the migration from OpenGL to X3D was surprisingly easy:
* writing X3D representation of all primitives used by pycam was very simple
* loading the resource html file containing the X3D in the webkit widget was also trivial

But there is a good amount of details left for bringing it back to the previous level of comfort:
* the html and x3d files should be provided directly by the pycam process (via http over a local port) instead of using a local directory
* a bit of javascript needs to be written for changing the viewport via gtk controls and for reloading the representation after changes (currently this resets the viewport)
* the required external javascript resource ([x3dom](https://x3dom.org/) - GPL2 or later) will need to be downloaded once and cached locally for offline usage

The current state is available in the branch `replace-opengl-with-x3d` - but it is surely only suitable for the daring souls ...