SebKuzminsky/pycam #137: Cant open simple 2d DXF

luw, 2019-01-07:
Hello, I am just trying to make some holes and slots in piece of plastic (2.5d operation).  DXF was created in autocad and saved in the format "AutoCAD 2010/LT2010 DXF (*.dxf)".  I have tried "AutoCAD R12/LT2 DXF" and had same issue.  The DXFs are attached in the zip.

Here is the error I received.
'''
An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot!
Traceback (most recent call last):
  File "/home/dave/Desktop/pycam-0.6.2/pycam/Gui/Project.py", line 386, in gui_activity_guard_wrapper
    result = func(self, *args, **kwargs)
  File "/home/dave/Desktop/pycam-0.6.2/pycam/Gui/Project.py", line 534, in load_model_file
    callback=progress.update)
  File "/home/dave/Desktop/pycam-0.6.2/pycam/Importers/DXFImporter.py", line 881, in import_model
    callback=callback)
  File "/home/dave/Desktop/pycam-0.6.2/pycam/Importers/DXFImporter.py", line 118, in __init__
    self.parse_content()
  File "/home/dave/Desktop/pycam-0.6.2/pycam/Importers/DXFImporter.py", line 241, in parse_content
    self.parse_lwpolyline()
  File "/home/dave/Desktop/pycam-0.6.2/pycam/Importers/DXFImporter.py", line 410, in parse_lwpolyline
    self.lines.extend(pycam.Geometry.get_bezier_lines(
AttributeError: 'module' object has no attribute 'get_bezier_lines
'''

Any ideas?

sumpfralle, 2019-01-07:
This should be fixed in the more recent release v0.6.3.
Maybe you overlooked it in the [list of releases](https://github.com/SebKuzminsky/pycam/issues/137), since it did not contain a manual description, thus it was formatted with a tiny font. I improved this now.

Please report back, whether your model loads successfully with v0.6.3. Otherwise: please attach the file :)

sumpfralle, 2019-02-13:
Assuming it is fixed in v0.6.3.

closed 2019-02-13 by sumpfralle