SebKuzminsky/pycam #138: Error pycam start

valeriob01, 2019-02-06:
~/pycam/pycam$ ./run_gui.py 
Enabled 4 parallel local processes
Storing a state revision (1/20)
Font directory: /home/valerio/pycam/share/fonts
Traceback (most recent call last):
  File "./run_gui.py", line 318, in <module>
    main_func()
  File "./run_gui.py", line 306, in main_func
    exit_code = execute(parser, args, pycam)
  File "./run_gui.py", line 230, in execute
    show_gui(workspace_filename=args.workspace_filename)
  File "./run_gui.py", line 112, in show_gui
    plugin_manager.import_plugins()
  File "/home/valerio/pycam/pycam/Plugins/__init__.py", line 273, in import_plugins
    self._load_plugin(plugin, filename, name)
  File "/home/valerio/pycam/pycam/Plugins/__init__.py", line 296, in _load_plugin
    self.enable_plugin(plugin_name)
  File "/home/valerio/pycam/pycam/Plugins/__init__.py", line 317, in enable_plugin
    plugin.enabled = plugin.setup()
  File "/home/valerio/pycam/pycam/Plugins/WebKitVisualization.py", line 40, in setup
    get_x3d_handler(self.core.get("get_x3d_export")))
  File "/home/valerio/pycam/pycam/Utils/httpserver.py", line 196, in create_with_random_port
    return cls((host, port), handler)
  File "/home/valerio/pycam/pycam/Utils/httpserver.py", line 172, in __init__
    self.httpd = http.server.ThreadingHTTPServer(address, handler)
AttributeError: module 'http.server' has no attribute 'ThreadingHTTPServer'

sumpfralle, 2019-02-06:
Uh - indeed I picked a recently added feature of that standard library module: it requires python 3.7 :(

I will fix that.

sumpfralle, 2019-02-13:
Fixed in e63aae44.

closed 2019-02-13 by sumpfralle

valeriob01, 2019-02-13:
It is starting now, but still it requires MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"

sumpfralle, 2019-02-14:
Where did you notice a reference to OpenGL?

The new code just embeds a webkit widget into the GTK window. I cannot imagine, how this could fail.

valeriob01, 2019-02-14:
The webkit widget just shows up a blank window without MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"

sumpfralle, 2019-02-14:
What a pity :(

I assume that your host is running all other OpenGL-related applications without problems?

valeriob01, 2019-02-14:
Not sure to catch all the humour here. Pycam is the only one application I run.

sumpfralle, 2019-02-14:
> Not sure to catch all the humour here.

Same for me :)

For clarification: I did not intend to pour humour into my previous question.

Are you running a virtualization setup or something similar? I am really confused, that something as widespread as webkit (used in modern browsers) could fail to work properly with OpenGL. But on the other hand, there are probably just too many layers involved ...

valeriob01, 2019-02-14:
No. I run Debian and everything else is Ok.

valeriob01, 2019-02-14:
> > Not sure to catch all the humour here.
> 
> Same for me :)
> 
> For clarification: I did not intend to pour humour into my previous question.

So that's why I didn't catch it at first :-)

sumpfralle, 2019-02-14:
This is a really confusing situation. I hoped to evade any potential client-specific issues by using a widespread implementation :(

I opened a separate issue for this problem: #139

valeriob01, 2019-02-15:
Right now it is working straight. No need for the MESA stuff. The buttons on top of the window (center, top, etc.) do not work.

[image]

commit e63aae44 (2019-02-13): Add alternative pre-Python3.7 implementation for ThreadingHTTPServer

Closes: #138