SebKuzminsky/pycam #139: webkit visualization: blank screen without MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"

sumpfralle, 2019-02-14:
See #138: the new webkit-based visualization (branch `replace-opengl-with-x3d`) shows a blank screen on a specific system. Setting the environment variable `MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"` works around this issue.

valeriob01, 2019-02-15:
Right now it is working straight. No need for the MESA stuff. The buttons on top of the window (center, top, etc.) do not work.

[image]

sumpfralle, 2019-02-15:
I am glad to hear this and I hope, such issues will not haunt us anymore ...

Regarding the buttons: yes, indeed they are not working at the moment. This is the last blocker for the switch to this visualization, I think.

closed 2019-02-15 by sumpfralle