SebKuzminsky/pycam #142: Access to API

Overdrivr, 2019-06-28:
I'm a bit confused by the name of the program, I was expecting a python library to enable fully-automated assignment of CAM operations and toolpath generation, but the doc seems to mention exclusively the use of the interface.

Given a 3D model such as STL or OBJ, is it possible to write a python script capable to interact with pycam without opening an interface ? Sorry if this is mentionned somewhere in the docs and I have missed it, but I've searched for a while and found nothing.

sumpfralle, 2019-07-28:
Indeed the current state of pycam in the `master` branch allows scripted (non-interactive) usage of PyCAM. It is not released, yet, but should be quite usable.

The documentation is lacking for now. Feel free to take a good look at the [example of a processing description](https://github.com/SebKuzminsky/pycam/blob/master/yaml_flow_working.yml).

Have fun exploring the details!

sumpfralle, 2019-07-28:
kk

Overdrivr, 2019-07-29:
@sumpfralle Thanks for the reply, that's not exactly what I was looking for, I was more hoping to get access to the python API to build cam operations programmatically. For complex workflows, YAML won't be enough.

sumpfralle, 2019-07-29:
At the moment, there is no HTTP-based API (if this is what you are looking for). Eventually it will be there, but please do not hold your breath ...

> For complex workflows, YAML won't be enough.

My reply was obviously way too short - sorry!

The current yaml-based format contains multiple collections: `models`, `tools`, `processes`, `bounds`, `tasks`, `toolpaths` and `export_settings`. I imagine, that these collections will be available in a REST-like API. The only other remaining detail described in the yaml format is `exports`. I imagine the available means of export to be represented as `POST` requests for generating output file (e.g. previews or 3D representations of the toolpath). From my point of view, this will be sufficient for building web-based applications with pycam as the CAM processor.

Back to the current yaml format: please understand it as the current *poor man's* API endpoint. At the moment you can create a web application, that assembles a yaml file for pycam, then uploads that file to a web service (which hands it over to pycam) and receive the resulting previews, toolpaths and so, that can be visualized in the web application.
Is this somehow along the lines of your plans?

Please share your thoughts!

Overdrivr, 2019-07-30:
Thanks for the reply !

I think my ticket's title was rather confusing, sorry about it.
I did not meant an HTTP api, I don't think that's the role for pycam.
pycam could easily be integrated into existing webservers such as django instead.

Rather, I would like to generate toolpaths directly in python, without the need for the GUI or using config files. Overall, this is already implemented but I guess it just needs to be documented and maybe refactored a bit.

What I would like to do is something along those lines:

```
import pycam

path = [
  (1.0, 2.0, 0.0),
  (3.0, 5.0, 0.0),
  ...
]

surf = [
  (2.0, 3.0, -1.0),
  (2.0, 5.0, -1.0),
  ...
]

# Configure some tools
tool = pycam.tool.FlatMill(radius=4.0)

# Generate simple program with cutting, pocketing, using provided tool
cnc_program = pycam.contour2d(path, tool=tool)
cnc_program += pycam.pocket2d(path, z_step=0.1, tool=tool)

# export to gcode
pycam.save(cnc_program, postprocessor='grbl')
```

ebo, 2019-07-30:
On Jul 30 2019 12:37 AM, Rémi Bèges wrote:
> Thanks for the reply !
>
> I think my ticket's title was rather confusing, sorry about it.
> I did not meant an HTTP api, I don't think that's the role for pycam.
> pycam could easily be integrated into [...]

Once you have that functionality you can also take things like 
converting images to edges, and edges to paths.

sumpfralle, 2019-07-31:
> Rather, I would like to generate toolpaths directly in python, without the need for the GUI or using config files.

Indeed - this is a good point!

In fact, I did not focus on such a use of pycam up to now. I will take your hints (slowly) into account ...
(contributions are obviously also more than welcome)

Thank you!

lsabi, 2019-11-16:
Any news about the status of this issue? I'm interested as well in such a feature

sumpfralle, 2019-11-17:
> Any news about the status of this issue?

No progress here, at the moment. Thanks for raising your voice of interest.

NovaProtocol, 2025-01-08:
Hey, love the idea.

I'm making a thesis for mechanical engineering and i was wondering if this is possible now?

For context, I'm automating agricultural drones to automatically fertilize the field with fertilizer. I made a code that generates the paths and it's working fine albeit a bit buggy on complex shapes. I was wondering if this feature is implemented so i can maybe export the gps data into an svg and somehow ask the api to path it for me?

If anyone also have suggestions that already solved my problem, I would appreciate if you would share them.

SebKuzminsky, 2025-01-08:
This might do what you're looking for: https://github.com/SebKuzminsky/svg2gcode