SebKuzminsky/pycam #147: Waterline doesn't make smooth cuts

yerazunis, 2019-08-23:
Waterline finishing occsionally inserts a "step" in the surface.    This is a different bug than #144  or #131.   

How to reproduce:

1. Use that same male bending die model:
[bender_die_v2_top.zip](https://github.com/SebKuzminsky/pycam/files/3535548/bender_die_v2_top.zip)

2. Create a 6.35mm roundnose tool 

3.  Create a process to do waterline milling  (you can switch it temporarily to slice removal to set "conventional" milling, as bug 131 suggests), Use 1mm stepdown.

4.  Go to "Tasks" and run the waterline milling withh the roundnose tool against the bender die model.

5.  You will get a result like this.   Note the "step" in the surface where no such step exists in the model.   Switching the model on and off helps you see it.

[image]

6.   The bug is not present in "profile" processing, which looks outstanding, nor in "slice removal" modes.  

[image]

yerazunis, 2019-08-26:
Hold off on debugging this - there is an interaction between how "overlap" and "stepdown" work  in "Waterline" mode that may turn this bug into PEBCAK or at least something that is much easier to fix!
      - Bill

ebo, 2019-08-26:
Since I had never heard/read the acronym PEBCAK before, I had to look 
it up ;-)

That said, if this is easily done we need to know how and at least 
document it (possibly stackoverflow style) or possibly capture it in a 
warning/error statement.

On Aug 26 2019 6:48 AM, yerazunis wrote:
> Hold off on debugging this - there is an interaction between how
> "overlap" and "stepdown" work  in "Waterline" mode that may turn this
> bug into PEBCAK or at least something that is much easier to fix!
>       - Bill

yerazunis, 2019-08-28:
Oh - the 0.25mm stepdown, 10% overlap, 6.37mm ballnose tool G-code actually violates the model (goes inside it - click for big image):

[image]

ebo, 2019-08-29:
Very nice torture test.  Actually, you might be able to make a smaller 
clipped version of this for a unit/regression test.  It is also good to 
catch the violation!

On Aug 28 2019 4:23 PM, yerazunis wrote:
> Oh - the 0.25mm stepdown, 10% overlap, 6.37mm ballnose tool G-code
> actually violates the model (goes inside it - click for big image):
>
> 
> [image]