SebKuzminsky/pycam #151: current master fail to run (SVG)

AndKe, 2019-10-16:
`~/Downloads/pycam-master/pycam$ ./run_gui.py 
Enabled 8 parallel local processes
Storing a state revision (1/20)
Font directory: /home/andre/Downloads/pycam-master/share/fonts
No preferences file found ([Errno 2] No such file or directory: '/home/andre/.pycam/preferences.conf'). Starting with default preferences.
Loading workspace from file: /home/andre/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
Traceback (most recent call last):
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/__init__.py", line 178, in <lambda>
    result = lambda *args: self.core.emit_event(func, *params)
  File "/home/andre/Downloads/pycam-master/pycam/Utils/events.py", line 136, in emit_event
    handler()
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/__init__.py", line 178, in <lambda>
    result = lambda *args: self.core.emit_event(func, *params)
  File "/home/andre/Downloads/pycam-master/pycam/Utils/events.py", line 136, in emit_event
    handler()
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/OpenGLViewDimension.py", line 51, in update_model_dimensions
    models = [m.get_model() for m in self.core.get("models").get_visible()]
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/OpenGLViewDimension.py", line 51, in <listcomp>
    models = [m.get_model() for m in self.core.get("models").get_visible()]
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 292, in wrapped
    return self.get_cached(inst, args, kwargs, calc_function)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 347, in get_cached
    cache_item = CacheItem(time.time(), calc_function(inst, *args, **kwargs))
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 1014, in get_model
    model = self.get_value("source").get(CollectionName.MODELS)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 292, in wrapped
    return self.get_cached(inst, args, kwargs, calc_function)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 347, in get_cached
    cache_item = CacheItem(time.time(), calc_function(inst, *args, **kwargs))
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 729, in get
    return self._get_source_location(source_type)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 256, in inner_function
    return func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 767, in _get_source_location
    detected_filetype = detect_file_type(location)
  File "/home/andre/Downloads/pycam-master/pycam/Importers/__init__.py", line 37, in detect_file_type
    from pycam.Importers.SVGDirectImporter import import_model as import_model_from_svg
  File "/home/andre/Downloads/pycam-master/pycam/Importers/SVGDirectImporter.py", line 33, in <module>
    import svg.path
ModuleNotFoundError: No module named 'svg'
Reverting collection changes due to error: No module named 'svg'
Imported 0 items into 'tools'
Imported 0 items into 'processes'
Imported 0 items into 'bounds'
Imported 0 items into 'tasks'
Imported 0 items into 'models'
Imported 0 items into 'toolpaths'
Imported 0 items into 'export_settings'
Imported 0 items into 'exports'
Traceback (most recent call last):
  File "./run_gui.py", line 340, in <module>
    main_func()
  File "./run_gui.py", line 328, in main_func
    exit_code = execute(parser, args, pycam)
  File "./run_gui.py", line 252, in execute
    show_gui(workspace_filename=args.workspace_filename)
  File "./run_gui.py", line 143, in show_gui
    gui.load_startup_workspace()
  File "/home/andre/Downloads/pycam-master/pycam/Gui/__init__.py", line 218, in load_startup_workspace
    default_content=DEFAULT_WORKSPACE)
  File "/home/andre/Downloads/pycam-master/pycam/Gui/__init__.py", line 249, in load_workspace_from_file
    return self.load_workspace_from_description(content)
  File "/home/andre/Downloads/pycam-master/pycam/Gui/__init__.py", line 290, in load_workspace_from_description
    reset=True)
  File "/home/andre/Downloads/pycam-master/pycam/Flow/parser.py", line 54, in parse_yaml
    if item_class(name, data) is None:
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 659, in __init__
    self.get_collection().append(self)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 564, in append
    self.notify_list_changed()
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 632, in notify_list_changed
    get_event_handler().emit_event(self._list_changed_event)
  File "/home/andre/Downloads/pycam-master/pycam/Utils/events.py", line 136, in emit_event
    handler()
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/__init__.py", line 178, in <lambda>
    result = lambda *args: self.core.emit_event(func, *params)
  File "/home/andre/Downloads/pycam-master/pycam/Utils/events.py", line 136, in emit_event
    handler()
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/OpenGLViewDimension.py", line 51, in update_model_dimensions
    models = [m.get_model() for m in self.core.get("models").get_visible()]
  File "/home/andre/Downloads/pycam-master/pycam/Plugins/OpenGLViewDimension.py", line 51, in <listcomp>
    models = [m.get_model() for m in self.core.get("models").get_visible()]
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 292, in wrapped
    return self.get_cached(inst, args, kwargs, calc_function)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 347, in get_cached
    cache_item = CacheItem(time.time(), calc_function(inst, *args, **kwargs))
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 1014, in get_model
    model = self.get_value("source").get(CollectionName.MODELS)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 292, in wrapped
    return self.get_cached(inst, args, kwargs, calc_function)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 347, in get_cached
    cache_item = CacheItem(time.time(), calc_function(inst, *args, **kwargs))
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 729, in get
    return self._get_source_location(source_type)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 237, in inner_function
    result = func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 256, in inner_function
    return func(self, *args, **kwargs)
  File "/home/andre/Downloads/pycam-master/pycam/workspace/data_models.py", line 767, in _get_source_location
    detected_filetype = detect_file_type(location)
  File "/home/andre/Downloads/pycam-master/pycam/Importers/__init__.py", line 37, in detect_file_type
    from pycam.Importers.SVGDirectImporter import import_model as import_model_from_svg
  File "/home/andre/Downloads/pycam-master/pycam/Importers/SVGDirectImporter.py", line 33, in <module>
    import svg.path
ModuleNotFoundError: No module named 'svg'

`

SebKuzminsky, 2019-10-16:
Looks like you don't have the `svg.path` python module installed.  If you're running Debian or Ubuntu, try running `sudo apt install python3-svg.path`.

SebKuzminsky, 2019-10-16:
Ah, at the top of the `INSTALL.md` file there's a list of packages you have to install to run pycam:

https://github.com/SebKuzminsky/pycam/blob/master/INSTALL.md

AndKe, 2019-10-17:
Worked, thank you. 
Actually, what threw me off was the fact that running run_gui very nicely suggested to install python3-opengl   - I did , and somehow "expected" an equally verbose suggestion if the failure was due to unmet dependencies.

closed 2019-10-17 by AndKe

sumpfralle, 2019-10-18:
> Actually, what threw me off was the fact that running run_gui very nicely suggested to install python3-opengl - I did , and somehow "expected" an equally verbose suggestion if the failure was due to unmet dependencies.

hehe - your expections were spoiled :)

Now I added a comparable warning message for this dependency. (see cc9a12d3c2cbca3ce6ff41f8867ba0db9f401b16).