SebKuzminsky/pycam #153: Exceptions on launch

jayMcBee, 2019-11-19:
Error shown when launching pycam:

An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot!
Traceback (most recent call last):
  File "/home/user/Things/pycam-0.6.4/pycam/Gui/Project.py", line 386, in gui_activity_guard_wrapper
    result = func(self, *args, **kwargs)
  File "/home/user/Things/pycam-0.6.4/pycam/Gui/Project.py", line 524, in load_model_file
    detected_filetype = pycam.Importers.detect_file_type(filename)
  File "/home/user/Things/pycam-0.6.4/pycam/Importers/__init__.py", line 38, in detect_file_type
    from pycam.Importers.SVGDirectImporter import import_model as import_model_from_svg
  File "/home/user/Things/pycam-0.6.4/pycam/Importers/SVGDirectImporter.py", line 36, in <module>
    import svg.path
ImportError: No module named svg.path

august9999, 2019-11-20:
i have installed python-svg.path and it solved

jayMcBee, 2019-11-20:
Thanks, that helped.
Gotta brush up my Unix/Linux skills after a break!

BTW, the dependencies appear to be incomplete - I had to install a couple of additional packages in order to launch pycam w/o exceptions

ArmoredBlood, 2019-12-22:
As another data point, I ran into the same issues on initial startup.

To resolve, I needed to run:
`sudo apt-get install python-gtkgext1 python-svg.path`.

[image]