SebKuzminsky/pycam #162: screenshots link not working

valeriob01, 2020-02-18:
An error has been encountered in accessing this page.

1. Server: pycam.sourceforge.net
2. URL path: /screenshots.md
3. Error notes: NONE
4. Error type: 404
5. Request method: GET
6. Request query string: NONE
7. Time: 2020-02-18 05:32:03 UTC (1582003923)

Reporting this problem: The problem you have encountered is with a project web site hosted by SourceForge.net. This issue should be reported to the SourceForge.net-hosted project (not to SourceForge.net).

If this is a severe or recurring/persistent problem, please do one of the following, and provide the error text (numbered 1 through 7, above):

    Contact the project via their designated support resources.
    Contact the project administrators of this project via email (see the upper right-hand corner of the Project Summary page for their usernames) at user-name@users.sourceforge.net

If you are a maintainer of this web content, please refer to the Site Documentation regarding web services for further assistance.

NOTE: As of 2008-10-23 directory index display has been disabled by default. This option may be re-enabled by the project by placing a file with the name ".htaccess" with this line:

Options +Indexes

valeriob01, 2020-03-11:
Lack of maintenance is the way to project failure.

sumpfralle, 2020-03-12:
> Lack of maintenance is the way to project failure.

Indeed. Just as a lack of contribution ...

I can assure you, that contributions (e.g. pull requests) will be handled swiftly and with pleasure. Thank you!

valeriob01, 2020-03-12:
> > Lack of maintenance is the way to project failure.
> 
> Indeed. Just as a lack of contribution ...
> 
> I can assure you, that contributions (e.g. pull requests) will be handled swiftly and with pleasure. Thank you!

This means relaunching the ball and not taking responsibility for your own project.