SebKuzminsky/pycam #164: Setting a tool length

DodoLaSaumure, 2020-03-19:
Dear all

Thank you very much for this sotware, I appreciate it very much.
I would like to suggest an improvement in tools management : 
I was wandering if this would be possible to set a tool length, in order to avoid collisions between the remaining material and the milling head.

Many thanks again to the developers and maintainers.
Best regards
Pierre

sumpfralle, 2020-03-20:
Do you mean a collision between the clamp (or spindle) and the material? This would be quite a challenging change, since it would require a 3D model of the related parts. Probably it could be reduced to a set of cylinders of varying diameter and height. The collision detection itself should be quite easy. But a usable GUI for such a specification feels challenging to me. Do you have an idea, how to accomplish such an interface?

Or did you mean something else?

DodoLaSaumure, 2020-03-20:
Thank you very much @sumpfralle.
Yes I mean a collision between the spindle and the remaining material.
But as you say, a simple set of two cylinders or a cylinder and a plane can be enough, I was not thinking of designing a 3D model of the spindle, just setting a length of a tool and a diameter for the spindle.
Anyway this is just an idea of improvment that I wanted to share with you.
Thanks  a lot
Have a good day
Best regards
Pierre

ebo, 2020-03-20:
I got to thinking about this.  I know that tool-length compensation is 
part of most CNC controller software (including LinuxCNC and 
MachineKit).  I wonder if there is some way to either input it up front 
(as a parameter/file/CMDL), or injest it from an external source (LCNC's 
Tool Table).

That said, it does not get to the real hart of the issue which is 
crashing the head/tool into the bed/head/fixture.

On Mar 20 2020 1:43 AM, Pierre Klein wrote:
> Thank you very much @sumpfralle.
> Yes I mean a collision between the spindle and the remaining
> material.
> But as you say, a simple set of two cylinders or a cylinder and a
> plane can be enough, I was [...]

sumpfralle, 2020-03-20:
> I know that tool-length compensation is part of most CNC controller software (including LinuxCNC and MachineKit).

This is probably the setting used by `pycam` for its touch-off operation? To be specific: moving the tool to a specific x/y location and going down until a switch is triggered. But (as far as I understand it) only relates to the tool length (no other obstacles).

> That said, it does not get to the real hart of the issue which is crashing the head/tool into the bed/head/fixture.

Yes, at least for me collisions with the fixture are the most common problem. This would be *really* hard to model, I guess.