SebKuzminsky/pycam #165: No module named gtk.

kim-rok, 2020-04-09:
kim@kim:~/pycam$ scripts/pycam

Enabled 2 parallel local processes
Failed to import GTK: No module named gtk
PyCAM dependency problem
Error: Failed to load the GTK interface.
Details:
	Python bindings for GTK+: MISSING (Install the package 'python-gtk2')

Detailed list of requirements: http://sf.net/apps/mediawiki/pycam/?title=Requirements

kim@kim:~$ sudo apt install python3-yaml gir1.2-gtk-3.0

Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-yaml is already the newest version (3.12-1build2).
python3-yaml set to manually installed.
gir1.2-gtk-3.0 is already the newest version (3.22.30-1ubuntu4).
The following packages were automatically installed and are no longer required:
  efibootmgr libfwup1 libwayland-egl1-mesa
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

NikSays, 2020-04-16:
So, error message suggests you to install package 'python-gtk2' did you run `sudo apt install python-gtk2`?

dlarue, 2020-05-12:
yes, does not solve the problem.  No documentation I've found yet works on my Ubuntu 18.04.3

valeriob01, 2020-05-12:
try with install the python-gtk2-dev version.

dlarue, 2020-05-12:
found a mention of installing pygobject.  Started from scratch by deleting my environment and removed --system-site-packages.

This seemed to work:

virtualenv -p python3 envCAM
 source envCAM/bin/activate
 pip install pygobject
 cd pycam
 pip install -r requirements.txt 
 pycam/run_gui.py

leekk4362, 2020-06-09:
I got the same problem with "Error: Failed to load the GTK interface." trying to install Pycam 0.6.4 in Ubuntu 20.04 LTS.  The python-gtk2/or with -dev packages are no longer available.  Is there a solution to this?  I am unsuccessful with the virtualenv approach.

leekk4362, 2020-06-09:
Got it to work.
Had to find many of the old packages, download them to my hard drive and manually install with sudo apt install <package>

python-gtk2_2.24.0-5.1ubuntu2_amd64.deb
python-gtkglext1_1.1.0-9.1_amd64.deb
python-svg.path_2.1.1-2_all.deb

Plus

sudo apt-get install libcanberra-gtk-module
sudo apt-get install python-enum34*

Hopefully I have not missed any step.