SebKuzminsky/pycam #168: pycam: command not found

DanielSchimpf, 2020-04-15:
I have unpacked pycam-master in my home directory.  I can run the gui by moving into that directory and them using `pycam/run_gui.py`.

Next I wanted to use pycam through the command-line. But trying to use commands such as: 

`pycam --help`

i get the error:

`pycam: command not found`

I assume I must do more than unpack the pycam-master folder to use these commands. What do I have to do to use pycam through the command line?

Thanks!

NikSays, 2020-04-16:
To install pycam scripts in /usr/bin/ you can run `sudo ./setup.py install` in the folder where setup.py is located (project root) and restart your terminal. Then you can just run `pycam`
Keep in mind that if you change any code in pycam you need to run this command again for changes to take place

DanielSchimpf, 2020-04-22:
That worked, thank you!

closed 2020-04-22 by DanielSchimpf