SebKuzminsky/pycam #177: 'python-gtk2' error

shenko, 2020-10-28:
'python-gtk2' error when running your install on Ubuntu 20.04. because Ubuntu no longer has 'pip' by default.

sgofferj, 2020-12-15:
It goes further than that. 20.04 does not support Python 2.7. It comes with Python 3. PyGTK, however is deprecated and there's no version for Python 3 any more.

Here's some info on porting to the successor:
https://wiki.gnome.org/action/show/Projects/PyGObject/IntrospectionPorting?action=show&redirect=PyGObject%2FIntrospectionPorting

https://python-gtk-3-tutorial.readthedocs.io/en/latest/index.html

inboxpm, 2022-02-07:
In 2022, error the same, "Failed to import GTK: No module named 'gtk'". Version is 0.6.4, marked as latest release, trying under Linux Mint 20.3.
I see a many forks of this project, maybe, anyone made it runnable under Python 3?