SebKuzminsky/pycam #187: ModuleNotFoundError: No module named 'pycam.errors' FreeBSD12.2

PKhurjekar, 2021-03-24:


rewolff, 2021-09-19:
Ubuntu 20.04: having the same problem.

petterreinholdtsen, 2021-11-12:
What operations do you do to trigger this error?  Can you provide a backtrace?

rewolff, 2021-11-12:
The operation that triggers this is "starting the program". 

I think, the conclusion for me was: pycam requires python 3.x 
First install of pycam I had python 3.x installed. Then one day I wanted to use SOMETHING on my system that required python 2. Apt-get automatically recognized the dependency and  installed python 2.7 for me. 

Now dpkg gets told to install python-2, and helpfully installs the link /usr/bin/python to point at python 2. 

This makes the program you're working with at that moment function perfectly, but sets things up for failure later down the line when you are no longer aware you installed something a while back. 

Anyway...  Suggestion: 
* ask running python for its version nr, if not the required level, throw an error. 
* Create a small wrapper that will invoke the main thing explicitly using "python-3.7" as the interpreter (falling down to python-3.6, python-3 and python if those don't exist.

Python is a language that throws unusable error messages for end users when things go wrong. 

rm: file not found

means  I mistyped the file I wanted to remove. pycam -> page full of shit seems to mean: "this program requires python3 and you're running it under python 2". Windows users are used to memorizing "Did not get an IP address from the router" means: "There is a patch you forgot to install" or something like that. I'm a Unix / Linux user and I'm used to programs just telling me what the problem is.