SebKuzminsky/pycam #188: Can't get pycam to work whatever method is used

manu-p, 2021-05-06:
Hi, 
I've been trying to install pycam on a Ubuntu 20.04 the different ways found in the so called "Docs" and also with the specific deb package.
For the later, forget it, no way to install pycam with it, unless you want to spend some time trying to compensate what's not been done for it work properly...

Now, [here](http://pycam.sourceforge.net/installation/) you're supposed to find 2 different processes to get pycam on your computer, one for the latest release, another for the dev version. Let's give the latest release a try:
**Installing the latest release**
**Linux / \*BSD**
1. [download](https://sourceforge.net/projects/pycam/files/) the archive of your choice
2. install the [requirements](http://pycam.sourceforge.net/requirements/) for your system
3. run scripts/pycam in the PyCAM directory

When you get to the download page and click in [pycam](https://sourceforge.net/projects/pycam/files/pycam/) since it looks like it's the latest modified kit (not sure it's a good choice though...), as of 2017-03-19, you can see that 0.5.1 has been modified **after** 0.6.1... 

Got the 0.5.1. No `scripts/pycam` script! Bad choice, I guess.

Got the 0.6.1 and... found the `scripts/pycam` to run! 

Then checked the requirements, followed the apt commands suggested, fine! 
Just noticed this:

> If your distribution does not ship GTK v3.16 or newer (e.g. Debian Jessie contains only v3.14), then you need to use an older release of PyCAM. The older releases (e.g. v0.6.x) depend on Python2 and GTK2 - thus they should work well even with rather old distributions.

Not sure I understand it very well... cause I found no package after 0.6.1... So I'm a bit confused with this. OK, let's keep going on.

But:
```
Traceback (most recent call last):
  File "scripts/pycam", line 54, in <module>
    from pycam.Geometry import Box3D, Point3D
  File "/usr/local/bin/pycam-0.6.1/pycam/Geometry/__init__.py", line 26, in <module>
    import pycam.Utils.log
  File "/usr/local/bin/pycam-0.6.1/pycam/Utils/__init__.py", line 27, in <module>
    import urlparse
ModuleNotFoundError: No module named 'urlparse'
```
Right... tried to install urlparse:
```
ERROR: Could not find a version that satisfies the requirement urlparse (from versions: none)
ERROR: No matching distribution found for urlparse
```
Beginning to get bored a bit, I found [here](https://github.com/SebKuzminsky/pycam/issues/65) an issue for the same error, and in the end, confidently saying in nov. 2017: 
> I am confident, that the above issue was fixed in 1e1af2b and 0cbd8b2.

If you say so...

Then, since I'm not the type to give up easily:
```
python2 scripts/pycam
Enabled 8 parallel local processes
Failed to import GTK: No module named gtk
PyCAM dependency problem
Error: Failed to load the GTK interface.
Details:
	Python bindings for GTK+: MISSING (Install the package 'python-gtk2')

Detailed list of requirements: http://sf.net/apps/mediawiki/pycam/?title=Requirements
```
OK, but my system says python-gtk2 has no version to be installed on it (or something quite near, it's written in french, my language).

Then I tried the dev version, and got no `run_gui.py` script anywhere!

In some way, looks like no one wants you to give pycam a try.

So... do I drop it all and try to find another way to get a "toolpath generator" easier to install? 
Or else, what else can I do to get pycam work on this computer?

So far, this is what we call the obstacle course in my language.

Thanks for reading, and for the help to come, maybe.

selroc, 2021-05-06:
It has been a while since the PyCAM author was actively developing it, meanwhile python3 and GTK3 were released and no adjustment has been made on PyCAM. But nowadays the author seems untraceable, thus I think we are on our own to fix such issues.

selroc, 2021-05-06:
In the end you should be able to run scripts/run_gui.py with the latest version if I remember right.

manu-p, 2021-05-06:
Thanks @selroc for replying. At least there's still this github for the (potential) users to come and see this is a dead project.

No, no `pycam/run_gui.py` in the 0.6.1 release.

```
/usr/local/bin/pycam-0.6.1$ ls pycam/
Cutters    Geometry  Importers    __init__.pyc    PathProcessors  Plugins      Simulation  Utils
Exporters  Gui       __init__.py  PathGenerators  Physics         __pycache__  Toolpath
/usr/local/bin/pycam-0.6.1$ find . -name run_gui.py
```
Nothing.

So, I was just trying to find a 'toolpath generator' s/w, pycam sounded good on the paper. Any advice for another solution easy to install and use? I get dxf files, and before sending them to the windows driven CNC machine, which is not mine and not here, I'd like to see how it is compliant before...

Thanks.

selroc, 2021-05-06:
If you look in github it is there pycam/run_gui.py , you have to clone the git repository.
`git clone https://github.com/SebKuzminsky/pycam.git`

manu-p, 2021-05-06:
Are you sure about that?
```
/usr/local/bin# git clone https://github.com/SebKuzminsky/pycam.git
Clonage dans 'pycam'...
remote: Enumerating objects: 16643, done.
remote: Total 16643 (delta 0), reused 0 (delta 0), pack-reused 16643
Réception d'objets: 100% (16643/16643), 46.25 Mio | 3.23 Mio/s, fait.
Résolution des deltas: 100% (12758/12758), fait.
/usr/local/bin# ls pycam/
Changelog           Dockerfile   Makefile     pycam             requirements.txt  setup.py               yaml_flow_working.yml
CODE_OF_CONDUCT.md  docs         man          pyinstaller       samples           share
COPYING.TXT         INSTALL.md   MANIFEST.in  README.md         scripts           technical_details.txt
debian              LICENSE.TXT  mkdocs.yml   release_info.txt  setup.cfg         yaml_flow_example.yml
```
Still no `pycam/run_gui.py` script in there.

Thanks for your help @selroc but I'm afraid we're at an ending point regarding pycam... 
Need to find another s/w for this.

BTW, just took a look at http://linuxcnc.org/, look interesting!

manu-p, 2021-05-06:
Oooops! Yes, there is a ./pycam/run_gui.py script!

And guess what, it starts!

selroc, 2021-05-06:
Just cloned:
```

$ git clone https://github.com/SebKuzminsky/pycam.git
Cloning into 'pycam'...
remote: Enumerating objects: 16643, done.
remote: Total 16643 (delta 0), reused 0 (delta 0), pack-reused 16643
Receiving objects: 100% (16643/16643), 46.25 MiB | 18.23 MiB/s, done.
Resolving deltas: 100% (12758/12758), done.
[sel@centos8w ~]$ cd pycam/
[sel@centos8w pycam]$ ls
Changelog           COPYING.TXT  Dockerfile  INSTALL.md   Makefile  MANIFEST.in  pycam        README.md         requirements.txt  scripts    setup.py  technical_details.txt  yaml_flow_working.yml
CODE_OF_CONDUCT.md  debian       docs        LICENSE.TXT  man       mkdocs.yml   pyinstaller  release_info.txt  samples           setup.cfg  share     yaml_flow_example.yml
[sel@centos8w pycam]$ ls pycam/
Cutters  errors.py  Exporters  Flow  Geometry  Gui  Importers  __init__.py  PathGenerators  PathProcessors  Physics  Plugins  run_cli.py  run_gui.py  Simulation  Test  Toolpath  Utils  workspace

```

selroc, 2021-05-06:
of course you must install the requirements plus some dependencies not listed, but you can desume them from error messages.

closed 2021-05-07 by manu-p

inboxpm, 2022-02-07:
A Git version is also don't work! Here is an error message from it: 
`Enabled 2 parallel local processes
Storing a state revision (1/20)

(run_gui.py:9306): Gdk-ERROR **: 20:43:35.583: The program 'run_gui.py' received an X Window System error.
This probably reflects a bug in the program.
The error was 'GLXBadFBConfig'.
  (Details: serial 375 error_code 170 request_code 152 (GLX) minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
`
What can I do with this?