SebKuzminsky/pycam #189: Unable to download Windows exe from Source Forge

jalovisko, 2021-06-15:
Attempting to download the Windows executibles from Source Forge ([this link](http://pycam.sourceforge.net/download.html)) lands me on a 404 page with the following output:
```
An error has been encountered in accessing this page.

1. Server: pycam.sourceforge.net
2. URL path: /download.html
3. Error notes: NONE
4. Error type: 404
5. Request method: GET
6. Request query string: NONE
7. Time: 2021-06-15 03:25:15 UTC (1623727515)

Reporting this problem: The problem you have encountered is with a project web site hosted by SourceForge.net. This issue should be reported to the SourceForge.net-hosted project (not to SourceForge.net).

If this is a severe or recurring/persistent problem, please do one of the following, and provide the error text (numbered 1 through 7, above):

Contact the project via their designated support resources.
Contact the project administrators of this project via email (see the upper right-hand corner of the Project Summary page for their usernames) at user-name@users.sourceforge.net
If you are a maintainer of this web content, please refer to the Site Documentation regarding web services for further assistance.

NOTE: As of 2008-10-23 directory index display has been disabled by default. This option may be re-enabled by the project by placing a file with the name ".htaccess" with this line:

Options +Indexes
```

selroc, 2021-06-15:
The main work happens on GitHub, you can get the latest version at https://github.com/SebKuzminsky/pycam/archive/refs/tags/v0.6.4.zip
or an older version at https://sourceforge.net/projects/pycam/files/pycam/0.6.1/pycam-0.6.1.zip/download

But there isn't any executable for Windows, PyCAM is written in Python which is an interpreted language.

jalovisko, 2021-06-15:
I see, thank you! I guess that I got confused by the Source Forge page [over here](http://pycam.sourceforge.net/installation/#windows) saying that I can `Use the standalone executable for Windows.` Should it be changed?

selroc, 2021-06-15:
Yes, but I am not the author. The author seems to have abandoned the project, but we are still able to use it "as is".