SebKuzminsky/pycam #211: Request for new project maintainers!

SebKuzminsky, 2024-05-25:
I don't have the bandwidth to maintain this project, and I haven't for a long time.

I'm happy to give maintainership to anyone who wants it, subject to community approval.

If you want to volunteer to be a PyCAM maintainer, post a reply in this issue announcing your availability.  You can optionally say a bit about yourself in your message. 

To vote for a new maintainer, add a :+1: emoji to their volunteering message.  To vote *against* a new maintainer, add a :-1: to their message.  In handful of days or so I'll give commit privilege to anyone with a few more ups than downs.

wosym, 2024-05-27:
I'm probably the worst candidate possible, since I'm very new in the world of CNC and have no clue how CAM software works. So any other would have my vote.

However, if there really are no other candidates, I could take up the task to at least take a look at the pull requests when they come in and merge them when they make sense. Because it would be a waste to let this beautiful project die due to being unmaintained. I'm pretty proficient in Git and Python.

SebKuzminsky, 2024-06-08:
I have invited @wosym to be a collaborator in the PyCAM repository.  Anyone else who wants to take on the burden of being a maintainer, DM me.

closed 2024-06-08 by SebKuzminsky